alibaba/AliOS-Things #1596: [Feature Request]: “MPU6050” is not accessed.

krisaaaaa, 2022-05-11:
### What do you want?

请问这个mpu6050是什么原因，导入不进来吗，怎么解决呢

[image]

### Version

master (Default)

skylarCai, 2022-05-11:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ethanlcz, 2022-05-11:
> ### Contact Details
> _No response_
> 
> ### What do you want?
> 请问这个mpu6050是什么原因，导入不进来吗，怎么解决呢
> 
> [image]
> 
> ### Version
> master (Default)

是怎么确认MPU6050没办法导入的呢？

ethanlcz, 2022-05-12:
> > > ### Contact Details
> > > _No response_
> > > ### What do you want?
> > > 请问这个mpu6050是什么原因，导入不进来吗，怎么解决呢
> > > [image]
> > > ### Version
> > > master (Default)
> > 
> > 
> > 是怎么确认MPU6050没办法导入的呢？
> 
> import后边的MPU6050是灰色的： [image] 请问这要怎么解决呢

from mpu6050 import MPU6050
MPU6050显示为灰色是VS Code检查到MPU6050这个类/函数你在代码中没有引用到，它可能认为对于你这个工程这句话是多余的。
你代码中有创建MPU6050的类对象并呼叫这个类提供的函数吗？

closed 2022-05-13 by krisaaaaa