alibaba/AliOS-Things #1597: [Bug Report]: esp32无法部署

wlxhl, 2022-05-11:
### What happened?

A bug happened!
运行环境:python
固件版本:2.1

### Version

master (Default)

### Relevant log output

```shell
esp32cam nodemcu-32s部署时问题
==== wait hardware reset!!! ====
==== hardware reset OK!!! ====
=== start upload... ===
=== wait python process interrupt ===
卡在这里不动了

vscode中显示下边这个
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x04034b50
invalid header: 0x04034b50
invalid header: 0x04034b50
invalid header: 0x04034b50

串口调试助手显示下边这个
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

skylarCai, 2022-05-11:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zetamind, 2022-05-13:
1、手动按板子上 reset 按键
2、连接串口或者串口调试助手查看程序运行状态是否正常
3、如正常再重新部署一次

wlxhl, 2022-05-14:
> 1、手动按板子上 reset 按键 2、连接串口或者串口调试助手查看程序运行状态是否正常 3、如正常再重新部署一次

按reset没有反应

wlxhl, 2022-05-16:
问题已解决，烧录固件时没解压，直接烧录zip文件。系统还显示烧录成功，建议优化一下这里，烧录时检查一下文件格式

closed 2022-05-16 by wlxhl

zetamind, 2022-05-16:
好的，感谢