alibaba/AliOS-Things #1610: [Bug Report]: Memory allocation failed 无限重启

hanfengsir, 2022-05-13:
### Contact Details

583187663@qq.com

### What happened?

NodeMCU-32S跑ADC采样实验或进行物联网平台连接时，无限重启
运行环境: 
固件版本: [HaaSPython-ESP32-v2.1.0] 

### Version

master (Default)

### Relevant log output

```shell
Memory allocation failed
Backtrace:...........xxx.xx.x.x.x.x.x.x.................
ELF file SHA256:00000000000000000000000
```

skylarCai, 2022-05-13:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yanxiaoyong-ch, 2022-05-14:
adc脚本执行之前，请执行下如下指令试试：
import kv
kv.set(''app_upgrade_channel', 'disable')

yanxiaoyong-ch, 2022-05-14:
自己的脚本逻辑执行之前，请执行下如下指令试试：
import kv
kv.set(''app_upgrade_channel', 'disable')

执行完后重启设备试试

YiluMao, 2022-05-14:
方便的话，能否把脚本贴出来，我们也测试下

hanfengsir, 2022-05-14:
> 自己的脚本逻辑之前，请执行下如下指令尝试执行： import kv kv.set(''app_upgrade_channel', 'disable')
> 
> 执行完后重启设备尝试

试过了，还是不行，HaaSPython-ESP32-v2.0.0运行是没有问题的，v2.1就无限重启，ADC实验的源码，无改动

hanfengsir, 2022-05-14:
> 方便的话，可以把脚本下贴出来，我们也可以

我直接copy官网案例上ADC实验源码，无改动的

hanfengsir, 2022-05-14:
> > 方便的话，可以把脚本下贴出来，我们也可以
> 
> 我直接copy官网案例上ADC实验源码，无改动的，HaaSPython-ESP32-v2.0.0运行是没有问题的，v2.1就无限重启

yanxiaoyong-ch, 2022-05-15:
https://haas.iot.aliyun.com/solution/detail/solution?versionId=800AED0564C26B6500000004&dataId=800AED0564C26B65 请问是这个体验案例吗

hanfengsir, 2022-05-15:
> https://haas.iot.aliyun.com/solution/detail/solution?versionId=800AED0564C26B6500000004&dataId=800AED0564C26B65请问是这个体验案例吗

是的没错，这个无法在V2.1上运行，无限重启，在v2.0上可以正常运行

yanxiaoyong-ch, 2022-05-16:
> 

我这不能复现你这个问题，麻烦将你出现问题的整个log 发一下

hanfengsir, 2022-05-16:
> > 
> 
> 我这不能复现你这个问题，麻烦将你出现问题的整个log 发一下
完整log如下：

E (9986) OTAPUT:: ===============HaaS token info===============

E (9986) OTAPUT:: HaaS device name: haas_94b555fc97f4

E (9996) OTAPUT:: device not activated

E (9996) OTAPUT:: =============================================

E (10006) OTAPUT:: starts connection with AIoT platform
establish mbedtls connection with server(host='a1ul4uS7RfV.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
wifi connected
establish tcp connection with server(host='h89m3EON3l1.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
tcp_connect: can only connect from state CLOSED
tcp_connect: can only connect from state CLOSED
success to establish tcp, fd=54
success to establish tcp, fd=55

Memory allocation failed

Backtrace:0x400915c2:0x3fffa600 0x40091af1:0x3fffa620 0x4017f672:0x3fffa640 0x40085537:0x3fffa660 0x400926dc:0x3fffa690 0x401d29b7:0x3fffa6d0 0x401012c1:0x3fffa710 0x400ff1cd:0x3fffa890

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffc678,len:4
load:0x3fffc67c,len:10788
load:0x40078000,len:12912
load:0x40080400,len:19980
entry 0x40080a00
test sram success
Apr 29 2022

Press key 'w' to 2ndboot cli menu in 100ms.
flash_get_info no:5 offset:0x16000 size:0x1000
128: ota crc cal:0x6514 param:0xffff
ota upg_flag:0xffffcount:0 crc:0xffffboot_type:255
No OTA upgrade.
mount fs
 ==== python execute bootpy ====
enable OneMinuteOnCloud
 ==== python file check /data/pyamp/main.py ====
 ==== python execute from /data/pyamp/main.py ====

PiaoLingAl, 2022-05-16:
> 

从这个log 上看，你的kv 并没有设置成功，请在系统开机以后， 按下 ctrl  +  c 打断程序，强制进入repl 模式，然后输入一下命令：
import kv
import kv
kv.set('app_upgrade_channel', 'disable')

hanfengsir, 2022-05-16:
> app_upgrade_channel

十分谢谢，已解决，但是我不明白，这个kv设置的作用是什么，我查了文档并找到没有相关答案？

YiluMao, 2022-05-17:
非常感谢您
您其实是发现了我们一个bug
我们会尽快修复这个问题，并发布新的固件

closed 2022-05-23 by skylarCai