alibaba/AliOS-Things #1612: 在VS中，新建项目，老板来了人脸识别案例编译有错误

gdfsytx, 2022-05-16:
### Contact Details

gdfsytx@yeah.net

### What happened?

开发板haas100,在VS中，使用C语言新建项目，老板来了人脸识别案例编译有错误

### Version

master (Default)

### What soultions are you seeing the problem on?

ucloud_ai_demo

### Relevant log output

```shell
> Executing task: aos make <

change dir to E:\rts\aiot\c\hass100\boss3\solutions\test_demo. 
save aos make params !!!
get current path
E:\rts\aiot\c\hass100\boss3\solutions\test_demo\package.yaml
test_demo
3
configuration haas100 written to E:\rts\aiot\c\hass100\boss3\.config. 
Build Solution by haas100 
scons: Reading SConscript files ...
There is no depend component `curl` found! 
========
[{'curl': 'master ? <!CHIP_HAAS1000>'}]
1
Reverse value
lost component check : no need install this component
[]
build toolchain checked =  arm-ali-aoseabi

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "E:\rts\aiot\c\hass100\boss3\solutions\test_demo\SConstruct", line 8, in <module>
install win_longcmd_spawn patch in windows. 
the script is E:\rts\aiot\c\hass100\boss3\hardware\chip\haas1000\_haas1000_alios_lds.py
current dir is E:\rts\aiot\c\hass100\boss3\hardware\chip\haas1000
toolchain is C:\Users\Administrator\.aliot\arm-ali-aoseabi\main\bin\arm-ali-aoseabi
cflag is -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -g -Os -DSECURE_BOOT -DHAAS_OTA_ENABLED=1 -DHAAS_OTA_BIN_VER=\"HAAS_AOS_3.1\" -DCHIP=haas1000 -DKERNEL=RHINO -DCHIP_HAAS1000_ARCH_ENHANCE -DCHIP_HAAS1000 -DMIC_SAMPLERATE=48000 -DMIC_PERIOD_MS=48 -DFLASH_SIZE=0x1000000 -DCRASH_DUMP_SECTION_SIZE=0 -DAUD_SECTION_SIZE=0 -DUSERDATA_SECTION_SIZE=0x1000 -DFACTORY_SECTION_SIZE=0x1000 -DNV_REC_DEV_VER=2 -DREVISION_INFO=x -DCHIP_HAS_UART=3 -DRESAMPLE_ANY_SAMPLE_RATE -DCHIP_HAS_TRANSQ -DRTOS=1 -DCQ_FUNC_ATTR= -DDEBUG=1 -DREL_TRACE_ENABLE -DLWIP_ETHERNETIF=1 -DLWIP_SUPPORT=0 -DPERSIST_DATA_SECTION_SIZE=0x1000 -std=gnu11 -fno-common -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -fomit-frame-pointer -DLASTWORD_RAM_ADDR=0x20166000 -DSYSINFO_PRODUCT_MODEL=\"ALI_AOS_HAAS100\"
cxxflag is -D_GLIBCXX_HAS_GTHREADS -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -DCHIP=haas1000 -DKERNEL=RHINO -DCHIP_HAAS1000_ARCH_ENHANCE -DCHIP_HAAS1000 -DMIC_SAMPLERATE=48000 -DMIC_PERIOD_MS=48 -DAUD_SECTION_SIZE=0 -DUSERDATA_SECTION_SIZE=0x1000 -DFACTORY_SECTION_SIZE=0x1000 -DFLASH_SIZE=0x1000000 -DCRASH_DUMP_SECTION_SIZE=0 -DSECURE_BOOT -DHAAS_OTA_ENABLED=1 -DHAAS_OTA_BIN_VER=\"HAAS_AOS_3.1\" -DNV_REC_DEV_VER=2 -DREVISION_INFO=x -DCHIP_HAS_UART=3 -DRESAMPLE_ANY_SAMPLE_RATE -DCHIP_HAS_TRANSQ -DRTOS=1 -DCQ_FUNC_ATTR= -DDEBUG=1 -DREL_TRACE_ENABLE -DLWIP_ETHERNETIF=1 -DLWIP_SUPPORT=0 -DPERSIST_DATA_SECTION_SIZE=0x1000 -DDISABLE_CLI_MULTI_CONSOLE=0 -std=gnu++11 -fsigned-char -fno-common -fno-exceptions -ffunction-sections -fdata-sections -fomit-frame-pointer -fpermissive -DSNTP_PERIODIC_SYNC_INTERVAL_MS=3600000 -g -Os
asmflag is -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -g -Os -DCHIP=haas1000 -DKERNEL=RHINO -DCHIP_HAAS1000_ARCH_ENHANCE -DCHIP_HAAS1000 -DMIC_SAMPLERATE=48000 -DMIC_PERIOD_MS=48 -c -x assembler-with-cpp -DLASTWORD_RAM_ADDR=0x20166000
ldflag is -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -nostartfiles --specs=nosys.specs -Wl,-Map=\\\"aos.map\\\" --coverage
cpu is cortex-m33
lib is E:\rts\aiot\c\hass100\boss3\solutions\test_demo\aos_sdk\lib\libhaas1000.a
solution_dir is E:\rts\aiot\c\hass100\boss3\solutions\test_demo
comp_name is haas1000
macro_list is -DAOS_COMP_ACTIVATION="y" -DCONFIG_NETMGR_WIFI_DEBUG=1 -DCONFIG_NETMGR_WIFI_LOG_LEVEL=4 -DNETMGR_SW_VERSION="netmgr_master" -DAOS_COMP_WIFI=1 -DWIFI_SW_VERSION="wifi_master" -DCONFIG_DRV_CORE=1 -DCONFIG_DRV_PLATFORM=1 -DCONFIG_DRV_VFS=1 -DCONFIG_COM_DRV=1 -DBASE_SW_VERSION="base_master" -DAOS_COMP_VFS=1 -DVFS_CONFIG_DEVICE_NODES=50 -DVFS_CONFIG_FD_OFFSET=48 -DVFS_CONFIG_PATH_MAX=256 -DVFS_CONFIG_MAX_FILE_NUM=50 -DVFS_CONFIG_CURRENT_DIRECTORY_ENABLE=1 -DRPM_INCLUDE_DIR="fs" -DVFS_CONFIG_DEBUG=1 -DVFS_SW_VERSION="vfs_master" -DPOSIX_SW_VERSION="posix_master" -DSYSINFO_SYSTEM_VERSION="1.0.3" -DAOS_KERNEL_BUILD=1 -DOSAL_RHINO=1 -DAOS_HAL=1 -DOSAL_AOS_SW_VERSION="osal_aos_master" -DAOS_COMP_DEBUG=1 -DDEBUG_PANIC_CLI=1 -DDEBUG_PANIC_CONTEXT_IN_STACK=1 -DDEBUG_CPUUSAGE_RECODE_TO_FILE_ENABLE=0 -DDEBUG_CPUUSAGE_MAX_TASK=80 -DDEBUG_CPUUSAGE_FILE_NAME="/data/cpuusage" -DDEBUG_LAST_WORD_ENABLE=0 -DDEBUG_LOG_FILE_NUM=10 -DDEBUG_LASTWORD_RAM_ADDR=0 -DDEBUG_LASTWORD_REGION_LEN=16384 -DDEBUG_LOG_DIR_NAME="/data/crash_reports" -DDEBUG_CONFIG_LOG_FILE_NAME="/data/crash_reports/crash_report" -DDEBUG_CONFIG_LOG_FILE_INDEX="/data/crash_reports/log_file_index" -DDEBUG_ULOG_FLUSH=0 -DDEBUG_SW_VERSION="debug_master" -DAOS_COMP_LITTLEFS=1 -DLFS_STATIC_OBJECT=1 -DLITTLEFS_FORMAT=1 -DLFS_CONFIG_TRACE=0 -DLFS_CONFIG_DEBUG=0 -DLFS_CONFIG_WARN=0 -DLFS_CONFIG_ERROR=1 -DLITTLEFS_SW_VERSION="littlefs_master" -DAOS_COMP_ULOG="y" -DULOG_CONFIG_ASYNC=0 -DULOG_CONFIG_POP_UDP=0 -DULOG_CONFIG_POP_FS=0 -DULOG_CONFIG_UPLOAD=0 -DULOG_CONFIG_STOP_FILTER_CLOUD=0 -DULOG_CONFIG_STOP_FILTER_UDP=4 -DULOG_CONFIG_STOP_FILTER_FS=4 -DULOG_CONFIG_STOP_FILTER_DEFAULT=0 -DULOG_CONFIG_SYSLOG_TIME_FORMAT=0 -DULOG_CONFIG_DEFAULT_DIR_ASYNC=0 -DULOG_CONFIG_LOG_SIZE=256 -DULOG_CONFIG_ASYNC_BUF_SIZE=6114 -DULOG_SW_VERSION="ulog_master" -DAOS_COMP_USERVICE=1 -DUSERVICE_SW_VERSION="uservice_master" -DSNTP_SW_VERSION="sntp_master" -DCONFIG_AOS_LWIP=1 -DCONFIG_TCPIP=1 -DWITH_LWIP_PKTPRINT=1 -DIPERF_ENABLED=1 -DLWIP_SW_VERSION="lwip_master" -DAOS_COMP_CLI=1 -DCLI_INBUF_SIZE=256 -DCLI_OUTBUF_SIZE=512 -DCLI_MAX_COMMANDS=128 -DCLI_MAX_ARG_NUM=16 -DCLI_MAX_ONCECMD_NUM=4 -DCLI_TASK_PRIORITY=60 -DCLI_CONFIG_STACK_SIZE=8192 -DCLI_TELNET_ENABLE=0 -DCLI_UAGENT_ENABLE=0 -DCLI_SEPRATED_CONSOLE=0 -DCLI_IOBOX_ENABLE=1 -DCLI_SW_VERSION="cli_master" -DLINKSDK_SW_VERSION="linksdk_master" -DAI_AGENT_SW_VERSION="ai_agent_master" -DCONFIG_ALICLOUD_FACEBODY_ENABLE=1 -DCONFIG_ALICLOUD_IMAGERECOG_ENABLE=1 -DCONFIG_ALICLOUD_IMAGEENHAN_ENABLE=1 -DCONFIG_ALICLOUD_OBJECTDET_ENABLE=1 -DCONFIG_ALICLOUD_IMAGESEG_ENABLE=1 -DCONFIG_ALICLOUD_OCR_ENABLE=1 -DCONFIG_ALICLOUD_CHATBOT_ENABLE=1 -DUSE_CRYPTO_MBEDTLS=1 -DOSS_ACCESS_KEY="Your-Access-Key" -DOSS_ACCESS_SECRET="Your-Access-Secret" -DOSS_ENDPOINT="Your-OSS-Endpoint" -DOSS_BUCKET="Your-OSS-Bucket" -DUCLOUD_AI_SW_VERSION="ucloud_ai_master" -DUSE_AOS_TIME_POSIX_API=1 -DOSS_DEBUG=0 -DREAD_SD_SIZE_MAX="1024*15" -DOSS_DISABLE_BUCKET=1 -DOSS_DISABLE_LIVECHANNEL=1 -DOSS_DISABLE_RESUAMABLE=1 -DOSS_DISABLE_ENCRYPTION=1 -DOSS_SW_VERSION="oss_master" -DCONFIG_HTTP_FILE_OPERATE=0 -DCONFIG_HTTP_SECURE=1 -DMBEDTLS_CONFIG_TLS_MAX_CONTENT_LEN=16384 -DHTTP_SW_VERSION="http_master" -DJSONCPP_SW_VERSION="jsoncpp_master" -DCONFIG_UGRAPHICS_FORMAT=0 -DCONFIG_UGRAPHICS_ROTATE=0 -DUGRAPHICS_SW_VERSION="ugraphics_master" -D__ALIOS__=1 -DLOAD_JPG=1 -DLOAD_BMP=1 -DLOAD_PNG=1 -D__native_client__=1 -DAOS_COMP_SDL2=1 -DSDL2_SW_VERSION="SDL2_master" -DFT2_BUILD_LIBRARY="y" -DFREETYPE_SW_VERSION="freetype_master" -DAOS_COMP_UDISPLAY=1 -DUDISPLAY_SW_VERSION="udisplay_master" -DFBDEV_SW_VERSION="fbdev_master" -DCONFIG_WIFI_CAMERA_ENABLE=1 -DWIFICAMERA_URL="http://192.168.43.166:80/capture" -DWIFICAMERA_FRAME_SIZE_CONTROL_URL="http://192.168.43.166:80/control\?var=framesize\&val=5" -DUCAMERA_SW_VERSION="ucamera_master" -DCONFIG_U_SPI_DEV=1 -DCONFIG_U_SPI_CORE=0 -DSPI_SW_VERSION="spi_master" -DAOS_COMP_GPIO=1 -DCONFIG_U_GPIO_DEV=1 -DCONFIG_U_GPIO_CORE=0 -DGPIO_SW_VERSION="gpio_master" -DCONFIG_CSI="csi2" -DCSI_SW_VERSION="csi_master" -DLIBC_STUB_SW_VERSION="libc_stub_master" -DCONFIG_U_I2C_DEV=1 -DCONFIG_U_I2C_CORE=1 -DCONFIG_U_I2C_WEAK_CSI=1 -DI2C_SW_VERSION="i2c_master" -DCONFIG_EPOLL_LWIP=1 -DCONFIG_WITH_MUSL=0 -DEPOLL_SW_VERSION="epoll_master" -DAOS_COMP_FLASH=1 -DCONFIG_U_FLASH_DEV=1 -DCONFIG_U_FLASH_CORE=1 -DFLASH_SW_VERSION="flash_master" -DAOS_COMP_UART=1 -DCONFIG_U_UART_DEV=1 -DCONFIG_U_UART_CORE=0 -DUART_SW_VERSION="uart_master" -DPWM_SW_VERSION="pwm_master" -DCONFIG_U_ADC_DEV=1 -DCONFIG_U_ADC_CORE=0 -DADC_SW_VERSION="adc_master" -DCONFIG_U_TIMER_DEV=1 -DCONFIG_U_TIMER_CORE=0 -DTIMER_SW_VERSION="timer_master" -DCONFIG_U_WDG_DEV=1 -DCONFIG_U_WDG_CORE=1 -DWATCHDOG_SW_VERSION="watchdog_master" -DINIT_SW_VERSION="init_master" -DAOS_COMP_KV=1 -DKV_SW_VERSION="kv_master" -DCONFIG_KERNEL_WORKQUEUE=1 -DCONFIG_SYSTICK_HZ=1000 -DCONFIG_KERNEL_RHINO=1 -DRHINO_CONFIG_MM_TLF=1 -DRHINO_CONFIG_MM_MINISIZEBIT=6 -DRHINO_CONFIG_MM_MAXMSIZE [... cut ...]
      |                    ^~~~~~
In file included from E:\rts\aiot\c\hass100\boss3\components\jsoncpp\include/json/json.h:11,
                 from E:\rts\aiot\c\hass100\boss3\components\ucloud_ai\src\model\aliyun-openapi\core\src\InstanceProfileCredentialsProvider.cc:22:
E:\rts\aiot\c\hass100\boss3\components\jsoncpp\include/json/reader.h:56:3: note: declared here
   56 |   Reader();
      |   ^~~~~~
scons: building terminated because of errors.
make: *** [all] Error 2
Build takes 0 Minutes 15 Seconds.

终端将被任务重用，按任意键关闭。
```

skylarCai, 2022-05-16:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

QingjiangXu, 2022-05-16:
详细的使用方法需要看案例里面的README.md文档。

比如下面的"<Your-Product-Key>"  需要替换成你自己的设备信息

/*开发者需要在此更换你的三元组信息：产品key、secret、设备名称、设备密钥*/
#define PRODUCT_KEY "<Your-Product-Key>"
#define PRODUCT_SECRET "<Your-Product-Secret>"
#define DEVICE_NAME "<Your-Device-Name>"
#define DEVICE_SECRET "<Your-Device-Secret>"

如果你再遇到问题，请继续回复

QingjiangXu, 2022-05-19:
> ### Contact Details
> [gdfsytx@yeah.net](mailto:gdfsytx@yeah.net)
> 
> ### What happened?
> 开发板haas100,在VS中，使用C语言新建项目，老板来了人脸识别案例编译有错误
> 
> ### Version
> master (Default)
> 
> ### What soultions are you seeing the problem on?
> ucloud_ai_demo
> 
> ### Relevant log output
> ```shell
> > Executing task: aos make <
> 
> change dir to E:\rts\aiot\c\hass100\boss3\solutions\test_demo. 
> save aos make params !!!
> get current path
> E:\rts\aiot\c\hass100\boss3\solutions\test_demo\package.yaml
> test_demo
> 3
> configuration haas100 written to E:\rts\aiot\c\hass100\boss3\.config. 
> Build Solution by haas100 
> scons: Reading SConscript files ...
> There is no depend component `curl` found! 
> ========
> [{'curl': 'master ? <!CHIP_HAAS1000>'}]
> 1
> Reverse value
> lost component check : no need install this component
> []
> build toolchain checked =  arm-ali-aoseabi
> 
> scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
> File "E:\rts\aiot\c\hass100\boss3\solutions\test_demo\SConstruct", line 8, in <module>
> install win_longcmd_spawn patch in windows. 
> the script is E:\rts\aiot\c\hass100\boss3\hardware\chip\haas1000\_haas1000_alios_lds.py
> current dir is E:\rts\aiot\c\hass100\boss3\hardware\chip\haas1000
> toolchain is C:\Users\Administrator\.aliot\arm-ali-aoseabi\main\bin\arm-ali-aoseabi
> cflag is -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -g -Os -DSECURE_BOOT -DHAAS_OTA_ENABLED=1 -DHAAS_OTA_BIN_VER=\"HAAS_AOS_3.1\" -DCHIP=haas1000 -DKERNEL=RHINO -DCHIP_HAAS1000_ARCH_ENHANCE -DCHIP_HAAS1000 -DMIC_SAMPLERATE=48000 -DMIC_PERIOD_MS=48 -DFLASH_SIZE=0x1000000 -DCRASH_DUMP_SECTION_SIZE=0 -DAUD_SECTION_SIZE=0 -DUSERDATA_SECTION_SIZE=0x1000 -DFACTORY_SECTION_SIZE=0x1000 -DNV_REC_DEV_VER=2 -DREVISION_INFO=x -DCHIP_HAS_UART=3 -DRESAMPLE_ANY_SAMPLE_RATE -DCHIP_HAS_TRANSQ -DRTOS=1 -DCQ_FUNC_ATTR= -DDEBUG=1 -DREL_TRACE_ENABLE -DLWIP_ETHERNETIF=1 -DLWIP_SUPPORT=0 -DPERSIST_DATA_SECTION_SIZE=0x1000 -std=gnu11 -fno-common -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -fomit-frame-pointer -DLASTWORD_RAM_ADDR=0x20166000 -DSYSINFO_PRODUCT_MODEL=\"ALI_AOS_HAAS100\"
> cxxflag is -D_GLIBCXX_HAS_GTHREADS -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -DCHIP=haas1000 -DKERNEL=RHINO -DCHIP_HAAS1000_ARCH_ENHANCE -DCHIP_HAAS1000 -DMIC_SAMPLERATE=48000 -DMIC_PERIOD_MS=48 -DAUD_SECTION_SIZE=0 -DUSERDATA_SECTION_SIZE=0x1000 -DFACTORY_SECTION_SIZE=0x1000 -DFLASH_SIZE=0x1000000 -DCRASH_DUMP_SECTION_SIZE=0 -DSECURE_BOOT -DHAAS_OTA_ENABLED=1 -DHAAS_OTA_BIN_VER=\"HAAS_AOS_3.1\" -DNV_REC_DEV_VER=2 -DREVISION_INFO=x -DCHIP_HAS_UART=3 -DRESAMPLE_ANY_SAMPLE_RATE -DCHIP_HAS_TRANSQ -DRTOS=1 -DCQ_FUNC_ATTR= -DDEBUG=1 -DREL_TRACE_ENABLE -DLWIP_ETHERNETIF=1 -DLWIP_SUPPORT=0 -DPERSIST_DATA_SECTION_SIZE=0x1000 -DDISABLE_CLI_MULTI_CONSOLE=0 -std=gnu++11 -fsigned-char -fno-common -fno-exceptions -ffunction-sections -fdata-sections -fomit-frame-pointer -fpermissive -DSNTP_PERIODIC_SYNC_INTERVAL_MS=3600000 -g -Os
> asmflag is -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -g -Os -DCHIP=haas1000 -DKERNEL=RHINO -DCHIP_HAAS1000_ARCH_ENHANCE -DCHIP_HAAS1000 -DMIC_SAMPLERATE=48000 -DMIC_PERIOD_MS=48 -c -x assembler-with-cpp -DLASTWORD_RAM_ADDR=0x20166000
> ldflag is -mthumb -mcpu=cortex-m33 -mfpu=fpv5-sp-d16 -mfloat-abi=hard -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -nostartfiles --specs=nosys.specs -Wl,-Map=\\\"aos.map\\\" --coverage
> cpu is cortex-m33
> lib is E:\rts\aiot\c\hass100\boss3\solutions\test_demo\aos_sdk\lib\libhaas1000.a
> solution_dir is E:\rts\aiot\c\hass100\boss3\solutions\test_demo
> comp_name is haas1000
> macro_list is -DAOS_COMP_ACTIVATION="y" -DCONFIG_NETMGR_WIFI_DEBUG=1 -DCONFIG_NETMGR_WIFI_LOG_LEVEL=4 -DNETMGR_SW_VERSION="netmgr_master" -DAOS_COMP_WIFI=1 -DWIFI_SW_VERSION="wifi_master" -DCONFIG_DRV_CORE=1 -DCONFIG_DRV_PLATFORM=1 -DCONFIG_DRV_VFS=1 -DCONFIG_COM_DRV=1 -DBASE_SW_VERSION="base_master" -DAOS_COMP_VFS=1 -DVFS_CONFIG_DEVICE_NODES=50 -DVFS_CONFIG_FD_OFFSET=48 -DVFS_CONFIG_PATH_MAX=256 -DVFS_CONFIG_MAX_FILE_NUM=50 -DVFS_CONFIG_CURRENT_DIRECTORY_ENABLE=1 -DRPM_INCLUDE_DIR="fs" -DVFS_CONFIG_DEBUG=1 -DVFS_SW_VERSION="vfs_master" -DPOSIX_SW_VERSION="posix_master" -DSYSINFO_SYSTEM_VERSION="1.0.3" -DAOS_KERNEL_BUILD=1 -DOSAL_RHINO=1 -DAOS_HAL=1 -DOSAL_AOS_SW_VERSION="osal_aos_master" -DAOS_COMP_DEBUG=1 -DDEBUG_PANIC_CLI=1 -DDEBUG_PANIC_CONTEXT_IN_STACK=1 -DDEBUG_CPUUSAGE_RECODE_TO_FILE_ENABLE=0 -DDEBUG_CPUUSAGE_MAX_TASK=80 -DDEBUG_CPUUSAGE_FILE_NAME="/data/cpuusage" -DDEBUG_LAST_WORD_ENABLE=0 -DDEBUG_LOG_FILE_NUM=10 -DDEBUG_LASTWORD_RAM_ADDR=0 -DDEBUG_LASTWORD_REGION_LEN=16384 -DDEBUG_LOG_DIR_NAME="/data/crash_reports" -DDEBUG_CONFIG_LOG_FILE_NAME="/data/crash_reports/crash_report" -DDEBUG_CONFIG_LOG_FILE_INDEX="/data/crash_reports/log_file_index" -DDEBUG_ULOG_FLUSH=0 -DDEBUG_SW_VERSION="debug_master" -DAOS_COMP_LITTLEFS=1 -DLFS_STATIC_OBJECT=1 -DLITTLEFS_FORMAT=1 -DLFS_CONFIG_TRACE=0 -DLFS_CONFIG_DEBUG=0 -DLFS_CONFIG_WARN=0 -DLFS_CONFIG_ERROR=1 -DLITTLEFS_SW_VERSION="littlefs_master" -DAOS_COMP_ULOG="y" -DULOG_CONFIG_ASYNC=0 -DULOG_CONFIG_POP_UDP=0 -DULOG_CONFIG_POP_FS=0 -DULOG_CONFIG_UPLOAD=0 -DULOG_CONFIG_STOP_FILTER_CLOUD=0 -DULOG_CONFIG_STOP_FILTER_UDP=4 -DULOG_CONFIG_STOP_FILTER_FS=4 -DULOG_CONFIG_STOP_FILTER_DEFAULT=0 -DULOG_CONFIG_SYSLOG_TIME_FORMAT=0 -DULOG_CONFIG_DEFAULT_DIR_ASYNC=0 -DULOG_CONFIG_LOG_SIZE=256 -DULOG_CONFIG_ASYNC_BUF_SIZE=6114 -DULOG_SW_VERSION="ulog_master" -DAOS_COMP_USERVICE [... cut ...]
> In file included from E:\rts\aiot\c\hass100\boss3\components\jsoncpp\include/json/json.h:11,
>                  from E:\rts\aiot\c\hass100\boss3\components\ucloud_ai\src\model\aliyun-openapi\core\src\InstanceProfileCredentialsProvider.cc:22:
> E:\rts\aiot\c\hass100\boss3\components\jsoncpp\include/json/reader.h:56:3: note: declared here
>    56 |   Reader();
>       |   ^~~~~~
> scons: building terminated because of errors.
> make: *** [all] Error 2
> Build takes 0 Minutes 15 Seconds.
> 
> 终端将被任务重用，按任意键关闭。
> ```

请问下您试的结果怎样，如果没有问题的话，我们就讲这个close了？

gdfsytx, 2022-05-22:
我是直接使用案例的，除了修改三元组信息，其他都没有修改，然后编译，但还是有上面的错误产生，你们自己生成一个老板来了的案例编译看看，是不是也有这个错误

QingjiangXu, 2022-05-23:
> 

给您带来困扰，深感歉意。
为了阻碍您的试用，麻烦手动注释掉一行代码。
hardware/chip/haas1000/package.yaml中189行
#  CONFIG_KWS_AI_ENGINE_ENABLE: 1   （前面加# 表示注释这行）

然后在下面点“清理”， 再点“编译”

skylarCai, 2022-06-08:
@gdfsytx

YiluMao, 2022-06-08:
@gdfsytx 
请问您的问题有没有被解决，如果没有进一步问题，请帮忙把问题close掉

closed 2022-06-13 by skylarCai