alibaba/AliOS-Things #1618: [Bug Report]: NodeMCU32设备断连WiFi，然后重新连接上之后，不会继续上传数据

hanfengsir, 2022-05-17:
### Contact Details

583187663@qq.com

### What happened?

A bug happened!
运行环境: NodeMCU32的ADC采样实验
固件版本: v2.1

### Version

master (Default)

skylarCai, 2022-05-17:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

ElliottSeer, 2022-05-17:
是自己推送的代码嘛 还是体验一分钟上云？

hanfengsir, 2022-05-17:
> 是自己推送的代码嘛 还是体验一分钟上云？

一分钟上云

hanfengsir, 2022-05-17:
我发现是：第二次连接上WiFi，回调成功的函数并没有触发，把devConnected的状态调整为True

ElliottSeer, 2022-05-17:
> > 是自己推送的代码嘛 还是体验一分钟上云？
> 
> 一分钟上云

目前一分钟上云的设计是这样的，如果设备因为关机或者断网离线之后，是不会主动重连的，只做体验使用。
如果你希望能够实现断线重连等功能，你需要自己实现相关的代码。

closed 2022-05-23 by skylarCai