alibaba/AliOS-Things #1637: 🚀🚀🚀 阿里云IoT | HaaS开源百校科技助力计划   -   招募校园大使！送HaaS EDU开发套件！

ElliottSeer, 2022-05-19:
👋 Hi，各位爱好科技的同学们

在这个技术快速升级、重构的时代

有这样一个活动，你不容错失

它就是**HaaS开源百校科技助力计划**

这是一项由**阿里云IoT HaaS技术团队**在全国高校推出的活动

帮助未来的新一代开拓技术视野

与同道中人开启技术路上的新机遇

如果你正好是这样的一名青年：

- [x] 乐于玩转物联网与嵌入式开发

- [x] 爱好探索技术前沿

- [x] 热爱折腾的科技Geek

- [x] 力图探寻物联网技术在未来无限 ♾️ 的可能性

**你，就是我们正在寻找的校园大使！**

在这个圈子里

你将和阿里云IoT以及全国各所高校的

同道中人、优秀创客、资深工程师直接对话

分享你的物联网项目和故事、阐述你的洞见

你还有机会与全国其他优秀的学生相识

利用HaaS开发框架的各项工具、资源

进行学习、训练与项目合作

获得更多的发展机会，更广袤的技术视野

一同以技术的名义，开启无边界的交流……

恰如旭日初升，科技的未来，一定属于你们！

## 关于HaaS开源百校科技助力计划

- HaaS开源百校科技助力计划面向国内高校科创组织和团体
- 旨在提供最前沿的产业物联网相关技术
- 促进开发人员交流，帮助学生专家与其他学生分享他们的故事和专业知识
- 作为校园大使，加入该计划的学生可以获得HaaS技术团队的技术支持和资源
- 利用HaaS开发框架的软件、工具和硬件开展工作

## 你能得到什么？

- **价值699元的HaaS开发套件**（非卖品），以及软件工具和丰富的案例库；
- 拥有**HaaS品牌纪念品**和**学生校园大使头衔**，正式与阿里云IoT HaaS技术团队建立关系；
- 直接对接HaaS技术团队的工程师和资源，优先获取最新的资讯、物料及最快速的技术支持；
- 不定期与HaaS技术团队进行线上或线下沟通，**有机会访问阿里巴巴园区及参加云栖大会**，与行业资深专家面对面交流；

## 你需要做些什么？

- 在一个及以上的校园科创组织或实验室中担任发言人或保持活跃；
- 使用HaaS开发框架尽情发挥你的创造力；
- 与HaaS技术团队一起，用技术和创新影响更多同学；

## 如何申请？

- 开源百校科技助力计划 2022 将招募100名校园大使，若无特殊情况，每所学校限一名。
- 计划于2022年 4月正式开始接受投递，截止至2022年9月结束投递。
- 在此期间，每接受到一个申请投递，HaaS技术团队将在一周内与申请人进行联系核实，并在每周三通知和公示结果。

[点击这里下载申请表文件](https://hli.aliyuncs.com/o/config/doc/HaaSCampusAmbassadorApplicationForm.docx)
请使用校园邮箱(.edu)将此申请表文件发送至 [haas-support@service.aliyun.com](mailto:haas-support@service.aliyun.com)

## 资源

[HaaS官方网站](https://haas.iot.aliyun.com/)

[HaaS学习中心](https://haas.iot.aliyun.com/learning)

[HaaS趣味案例](https://haas.iot.aliyun.com/solution)

[CSDN HaaS技术社区](https://blog.csdn.net/HaaSTech)

[image]

ethanlcz, 2022-05-19:
哇，好活动yeah

DKSCIcat, 2023-07-29:
现在还有啥活动吗🫡