alibaba/AliOS-Things #1665: m5 没有sd接口，自行查阅esp32的micropython文件。按文档操作还是挂载不了，麻烦请协助一下。

rualdeng, 2022-05-20:
目前M5Stack Core2上SD卡的功能没有调试，你遇到的问题我们会紧急处理一下。
建议你先用文件系统的方式操作，直接将数据放到/data分区下，等SD卡功能完成后再切换到SD卡上。

_Originally posted by @yong171966 in https://github.com/alibaba/AliOS-Things/issues/1465#issuecomment-1081817581_

axp = Axp192()
axp.setSpkEnable(1)
axp.setLDOVoltage(2,3300)
axp.setLDOEnable(2,True)
utime.sleep(1)

sd= machine.SDCard(slot=2,sck=Pin(18), mosi=Pin(23),miso=Pin(38), cs=Pin(4))

ret = uos.mount(sd, "/sd")

skylarCai, 2022-05-20:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

rualdeng, 2022-05-20:
打印sd.info()。可以获取sd卡信息，就是挂载不上，不知是什么原因，还请协助一下。谢谢

closed 2022-06-09 by skylarCai