alibaba/AliOS-Things #1719: [Bug Report]: 土壤湿度传感器描述不清楚

ethanlcz, 2022-05-25:
### 您发现了什么问题

https://haas.iot.aliyun.com/solution/detail/hardware?versionId=800CCF5AD5CBBCEA00000001&dataId=800CCF5AD5CBBCEA&s=
下图中的阈值是指什么？怎样调节阈值呢？文档中没有对怎样调节阈值进行描述。
[image]

### 分支

master (Default)

skylarCai, 2022-05-25:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zy-sunshine, 2022-05-25:
你好，软件无法设置阈值
这个传感器模块上有一个电位器，通过调节电位器调整DO的输出0 1 的阈值，是一个经验值
[image]

ethanlcz, 2022-06-07:
> 你好，软件无法设置阈值 这个传感器模块上有一个电位器，通过调节电位器调整DO的输出0 1 的阈值，是一个经验值 [image]

可以在文档中moistureDetect函数的说明中注释一下吗？不然会很疑惑。

ethanlcz, 2022-06-09:
传感器驱动文档中已经增加了对阈值进行调整方法的描述。
https://github.com/alibaba/AliOS-Things/pull/1820

closed 2022-06-09 by ethanlcz