alibaba/AliOS-Things #1720: [Bug Report]: 传感器上云时出现Memory Allocation failed

ethanlcz, 2022-05-25:
### 联系方式

277645743@qq.com

### 您发现了什么问题

A bug happened!
运行环境:
固件版本:HaaSPython-ESP32-v2.1.0

### 分支

master (Default)

### 日志

```shell
[Bug Report]:传感器上云时出现Memory allocation failed错误

277645743@qq.com
运行环境:VScode haas
固件版本:HaaSPython-ESP32-v2.1.0

Relevant log output ：
PS C:\Users\Administrator\workspace\HaaSPython> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.1.0\asserts\mpy_tools\mpytool.py -p COM8                 -f c:\Users\Administrator\workspace\HaaSPython\solutions\temphumi -t c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.1.0\asserts\mpy_tools
[*] port is  COM8
[*] Filename is  c:\Users\Administrator\workspace\HaaSPython\solutions\temphumi
[*] Filename is  c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.1.0\asserts\mpy_tools
C:\Users\Administrator\AppData\Roaming\Thonny
==== wait hardware reset!!! ====
==== hardware reset OK!!! ====
=== start upload... ===
=== wait python process interrupt ===
upload process...0%
upload process...9%
upload process...15%
upload process...22%
upload process...28%
upload process...34%
upload process...40%
upload process...52%
upload process...64%
upload process...70%
upload process...78%
upload process...90%
upload process...97%
=== upload progress end ===
6.627444267272949
5.456299781799316
--- Miniterm on COM8  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
flash_get_info no:5 offset:0x16000 size:0x1000
128: ota crc cal:0x6514 param:0xffff
ota upg_flag:0xffffcount:0 crc:0xffffboot_type:255
No OTA upgrade.
mount fs
 ==== python execute bootpy ====
enable OneMinuteOnCloud
 ==== python file check /data/pyamp/main.py ====
 ==== python execute from /data/pyamp/main.py ====
sht3x inited!
[   0.850]<E>DRIVER_I2C aos_hal_i2c_master_send failed
start to connect  BOHANLED
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
[... 177 lines cut ...]

E (9537) OTAPUT:: =============================================

E (9547) OTAPUT:: starts connection with AIoT platform
establish mbedtls connection with server(host='a1ul4uS7RfV.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
wifi_connected: False
('192.168.0.127', '255.255.255.0', '192.168.0.1', '192.168.1.1')
establish tcp connection with server(host='h56ni1WnZvY.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
sleep for 1 s
tcp_connect: can only connect from state CLOSED
success to establish tcp, fd=54
tcp_connect: can only connect from state CLOSED
success to establish tcp, fd=55
���联网平台连接成功
sleep for 2s

Memory allocation failed

Backtrace:0x400915c2:0x3ffdee60 0x40091af1:0x3ffdee80 0x4017f672:0x3ffdeea0 0x40085537:0x3ffdeec0 0x400857d1:0x3ffdeef0 0x40084e4c:0x3ffdef10 0x4016f711:0x3ffdef30 0x40170c2a:0x3ffdef50 0x40170cea:0x3ffdef80 0x4016e0a2:0x3ffdefb0 0x4016dd7d:0x3ffdeff0 0x4016dd9e:0x3ffdf030 0x4016df61:0x3ffdf060 0x4018122d:0x3ffdf0a0 0x4017bd30:0x3ffdf1b0 0x4017741f:0x3ffdf1f0 0x4017b1e8:0x3ffdf2a0 0x4017b20b:0x3ffdf2c0 0x401e1037:0x3ffdf2e0 0x401e10e8:0x3ffdf320 0x401db407:0x3ffdf360 0x401db994:0x3ffdf520 0x401db9f3:0x3ffdf550 0x401d1e25:0x3ffdf570

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57
```

skylarCai, 2022-05-25:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

wushengtws, 2022-05-25:
不操作传感器，直接模拟数据上传物联网平台会不会出现问题的呢

zhirencao, 2022-05-25:
> 不操作传感器，直接模拟数据上传物联网平台会不会出现问题的呢

   1. get_wifi_status()
    2.connect_lk(productKey, deviceName, deviceSecret)
    3.upload_temperature_and_Humidity()
单独运行1没问题，运行1和2关掉3有问题，也就是只要运行2就会有问题，操作传感器是在3里面

closed 2022-06-02 by ethanlcz