alibaba/AliOS-Things #1724: 🧑‍💻 🏕  🚀 Alibaba Summer of Code (ASOC) 2022

ElliottSeer, 2022-05-25:
# Alibaba Summer of Code (ASOC) 2022
Welcome to the open source world! If you haven't planned how to spend this summer, come to the Alibaba Summer of Code and code with us! 💻

Alibaba Summer of Code is a global program focused on engaging students directly in open source software development. Under the guidance of the mentor in the Alibaba open source project, students can experience software development in the real world. Alibaba Summer of code will begin from May 30th to September 1st. Students can use the summertime to participate in the open source project and work with the core members of the project.

This is a master issue to track the progress and result of [Alibaba Summer of Code 2022](https://opensource.alibaba.com/asoc2022).

# What you can get?
On this exclusive developer journey, students will have the opportunity to:

Participate in the top projects of the International Open Source Foundation;
Get a scholarship from Alibaba;
Obtain an open source contributor certificate;
Get a fast pass of Alibaba Internship
Get your code adopted and used by the open source project！

# Timeline
[image]

# Apply Now!
Browse open idea list here:

#1723 
#1722 
#1721 

# Contact the Organizer
If you have any questions, visit the event website: https://opensource.alibaba.com/asoc2022

Email address: [opensource@alibaba-inc.com](mailto:opensource@alibaba-inc.com)

skylarCai, 2022-05-25:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

SamanthaWangdl, 2022-06-09:
老师好，IoT的项目能给个联系的邮箱吗？

skylarCai, 2022-06-09:
@SamanthaWangdl 同学你好，可以通过如下两种方式联系我们哦

Email address: [opensource@alibaba-inc.com](mailto:opensource@alibaba-inc.com)

钉钉搜索群号：44561832