alibaba/AliOS-Things #1727: [Bug Report]: esp32 micropython环境，lvgl初始化后，无法使用网络接口

fumiot, 2022-05-25:
### 联系方式

qiugang@wuhanxl.com

### 您发现了什么问题

esp32 Python轻代码框架，lvgl初始化后，调用wlan = network.WLAN(network.STA_IF)会报Memory allocation failed.
运行环境:ESP32  m5stackcore2
固件版本:[HaaSPython-ESP32-8M-v2.1.0](https://hli.aliyuncs.com/o/config/HaaS_Python/HaaSPython-ESP32-8M-v2.1.0.zip)

### 分支

rel_3.3.0 (Main)

### 请选择您发现此问题的解决方案名称

py_engine_esp32_demo

### 日志

```shell
>>> import lvgl as lv
>>> from display import LVGL as disp
>>> import network
>>> lv.init()
E (3477) disp_lvgl: app_init
>>> disp.init()
E (3687) disp_lvgl: register touch driver
>>> wlan = network.WLAN(network.STA_IF)

Memory allocation failed

Backtrace:0x40093dd1:0x3ffea9c0 0x400943bd:0x3ffea9e0 0x401aac3a:0x3ffeaa00 0x40085d93:0x3ffeaa20 0x40081c2d:0x3ffeaa50 0x401dd730:0x3ffeaa70 0x401d301a:0x3ffeaab0 0x401c27fe:0x3ffeaad0 0x401c288f:0x3ffeaaf0 0x401bfd0e:0x3ffeab10 0x401da0e2:0x3ffeab30

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
```

skylarCai, 2022-05-25:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

yanxiaoyong-ch, 2022-06-08:
这周预计发布最新版本，会解决这个问题

YiluMao, 2022-07-13:
请问您还有问题不？

fumiot, 2022-07-14:
master分支下的page_solutions.py，无法识别车牌。运行该案例，只有图片显示，车牌号无法识别。
串口REPL持续输出：
tcp_connect: can only connect from state CLOSED
tcp_connect: can only connect from state CLOSED
tcp_connect: can only connect from state CLOSED