alibaba/AliOS-Things #1747: [Bug Report]: 重复显示Press key 'w' to 2ndboot cli menu in 100ms，无法连接网络

mxjoke, 2022-05-28:
### 联系方式

1097804505@qq.com

### 您发现了什么问题

A bug happened!
运行环境:
固件版本:v2.1.0
一开始使用智能刷卡门禁系统以及乐鑫开发板时能够直接成功联网用的是电脑的2.4g热点，但在后续操作中一直跳出2ndboot cli menu，在循环10-20s后结束开始联网，但始终无法连接；尝试过重新刷入固件，和更换开发样例，但都失败，该菜单始终存在，且联网始终失败

### 分支

rel_3.3.0 (Main)

### 请选择您发现此问题的解决方案名称

helloworld_demo

### 日志

```shell
Press key 'w' to 2ndboot cli menu in 100ms.
flash_get_info no:5 offset:0x16000 size:0x1000
128: ota crc cal:0x6514 param:0xffff
ota upg_flag:0xffffcount:0 crc:0xffffboot_type:255
No OTA upgrade.
mount fs
 ==== python execute bootpy ====
 ==== python file check /data/pyamp/main.py ====
 ==== python execute from /data/pyamp/main.py ====

Memory allocation failed

Backtrace:0x400915c2:0x3ffcfa50 0x40091af1:0x3ffcfa70 0x4019bfde:0x3ffcfa90 0x40085537:0x3ffcfab0 0x40085648:0x3ffcfae0 0x40095ab2:0x3ffcfb00 0x40095ad0:0x3ffcfb20 0x4011a087:0x3ffcfb40 0x400fe21e:0x3ffcfba0 0x401065c1:0x3ffcfbc0 0x401066ee:0x3ffcfbe0 0x40109315:0x3ffcfc00 0x400fe380:0x3ffcfca0 0x401065c1:0x3ffcfd00 0x4010926d:0x3ffcfd20 0x400fe380:0x3ffcfdc0 0x401065c1:0x3ffcfe20 0x401065ea:0x3ffcfe40 0x401171ef:0x3ffcfe60 0x40117569:0x3ffcfef0 0x401175cd:0x3ffcff10 0x400d2fca:0x3ffcff30

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffc678,len:4
load:0x3fffc67c,len:10772
load:0x40078000,len:12912
load:0x40080400,len:19980
entry 0x40080a00
test sram success
May 18 2022
```

skylarCai, 2022-05-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-05-29:
我们内部先分析下，感谢发现问题

mxjoke, 2022-05-30:
我周末在自己研究的时候看来下，是WiFi那部分的程序有问题，Memory allocation failed，我在网上找到的资料是有死循环的错误，但我看了下程序也没有问题，感觉是硬件有东西烧掉了，就想请问下有判断硬件坏掉的办法吗？稍微便宜点的板子有推荐吗？50块以上的板子有点贵了，毕竟这个大作业就用一次，其余的时候用不上。谢谢您。

抄送: "Mr .young ***@***.******@***.***&gt;;

yanxiaoyong-ch, 2022-05-30:
请问main.py做了什么事情，方便main.py贴出来看一下吗

mxjoke, 2022-05-30:
我查出来天线出了问题，捂脸，好的板子刚刚才到，重新试了下，都能够运行的，硬件的原因

抄送: "Mr .young ***@***.******@***.***&gt;;

yanxiaoyong-ch, 2022-05-30:
好的，那我们暂时关掉这个issue了

yanxiaoyong-ch, 2022-05-30:
好的，如果问题已经找到，帮忙关掉这个issue哈

closed 2022-05-30 by mxjoke