alibaba/AliOS-Things #1748: [Bug Report]: ds18b20驱动执行失败

15023019082, 2022-05-28:
### 联系方式

690482152@qq.com

### 您发现了什么问题

A bug happened!
运行环境:helloworld案例
固件版本:HaaSPython-HaaS200-v2.1.0

class DS18B20():
    def __init__(self, gpioObj, resolution=12):
        print("1")
        self.pin = None
        if not isinstance(gpioObj, GPIO):
            raise ValueError("parameter is not a GPIO object")

        self.pin = gpioObj.port()
        self.no_addr=0
        self.addr=self.getaddr()
        self.unit='c'
        self.res=resolution
        print("2")

    def getaddr(self):
        print("3")
        ow=OneWire(Pin(self.pin))
        print("4")
        print(ow)
        a=ow.scan()
        print("5") //这里没有执行
        for i in a:
            self.no_addr+=1
        print("6")
        return a

### 分支

master (Default)

### 请选择您发现此问题的解决方案名称

helloworld_demo

### 日志

```shell
PS E:\haas\project\wd\HaaSPython> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python -m serial.tools.miniterm COM5 115200 --raw
--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
mac = a4 1a 3a 57 26 08 ,       rssi = -41,     ssid = 27-7,    channel = 11,   sec = wpa/wpa2
wifi_start_sta:2206 start wifi

connect to ssid:27-7,password:./123456+-

joining BSS by BSSID a4:1a:3a:57:26:08 ...    

RTL8721D[Driver]: set BSSID: a4:1a:3a:57:26:08

RTL8721D[Driver]: set ssid [27-7] 

RTL8721D[Driver]: rtw_set_wpa_ie[1160]: AuthKeyMgmt = 0x2 

RTL8721D[Driver]: rtw_restruct_sec_ie[4202]: no pmksa cached 

RTL8721D[Driver]: start auth to a4:1a:3a:57:26:08

RTL8721D[Driver]: auth alg = 2

RTL8721D[Driver]:
OnAuthClient:algthm = 0, seq = 2, status = 0, sae_msg_len = 0

RTL8721D[Driver]: auth success, start assoc

RTL8721D[Driver]: association success(res=2)
wlan1: 1 DL RSVD page success! DLBcnCount:01, poll:00000001

RTL8721D[Driver]: ClientSendEAPOL[1600]: no use cache pmksa

RTL8721D[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8721D[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1

Wifi_connect successed
ssid_contain_chinese_check:2224 Error: app ssid is NULL!
set ip: 192.168.0.101
set mask: 255.255.255.0
set gw: 192.168.0.1
save wifi info to kv and start ota channel
[  13.092]<E>ACTIVATION_REPORT activation_report:288 report_data:{"activationStr":"V=3.3.0&O=Anolis Things&P=AliOS Things Team&A=HaaS Python&B=haas200&C=rtl872xd&N=WIFI/BT&X=HaaS Inside&S=MQTT/HTTPS&T=solo&M=9c-44-3d-56-d6-ee&Y=aliyun"}
[  13.115]<E>ACTIVATION_REPORT activation_report:292 report http data:POST /report HTTP/1.1
Host:os-activation.iot.aliyun.com
User-Agent: AliOS-Things
Content-Length:171
Accept: */*
Content-Type:application/json
Connection: Keep-Alive

{"activatio[  13.332]<E>ACTIVATION_REPORT socket connect success
[  13.934]<E>KV aos_kv_get failed key = _amp_pyapp_url;
[  14.194]<E>OTAPUT: ===============HaaS token info===============
[  14.200]<E>OTAPUT: HaaS device name: haas_9c443d56d6ee
[  14.208]<E>OTAPUT: HaaS token: haas_9c443d56d6ee&&abd34fd368b7ca27861fc32500dd1fcc
[  14.216]<E>OTAPUT: =============================================
establish mbedtls connection with server(host='a1ul4uS7RfV.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
gpio pin:3
1
3
gpio pin:3
4
<OneWire object at 20041c0>
```

skylarCai, 2022-05-28:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

jinliangli, 2022-05-31:
抱歉，这个ds18b20传感器在HaaS200上不支持。 在ESP32的开发板是支持的。

jinliangli, 2022-06-08:
你好， 如果没有其他问题，该问题将在明天关闭。

closed 2022-06-09 by skylarCai