alibaba/AliOS-Things #1751: [Bug Report]: HaaS 轻应用固件

krisaaaaa, 2022-05-29:
### What happened?

A bug happened!
运行环境:
固件版本:

### Version

master (Default)

### Relevant log output

```shell
请问HaaS Python 轻应用固件中包含了哪些内容呢，HaaS EDU K1。
```

skylarCai, 2022-05-29:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-06-08:
是支持EDU K1的
请问您还有什么问题？

krisaaaaa, 2022-06-08:
> 是支持EDU K1的 请问您还有什么问题？

我想知道这个固件里都封装了一些什么东西在里面呢

krisaaaaa, 2022-06-08:
> 主要是支持我们HaaS Python提供的API 您可以参考我们官网的文档： https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16145223.0.0.595660b1BeBimb#/

好的，谢谢

YiluMao, 2022-06-08:
如果您的问题解决了，帮忙把两个相关的问题都close掉，多谢

closed 2022-06-10 by skylarCai