alibaba/AliOS-Things #1762: [Bug Report]: haas studio 报错

chaojibinbin, 2022-06-02:
### What happened?

[*] port is  COM6
[*] Filename is  c:\code\haasesp32\HaaSPython\src
[*] Filename is  c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools
C:\Users\Administrator\AppData\Roaming\Thonny
==== wait hardware reset!!! ====
==== hardware reset OK!!! ====
Traceback (most recent call last):
  File "c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools\mpytool.py", line 112, in <module>
    file_transfer.push_testcase_to_devices_via_repl(file, port, tool)
  File "c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools\file_transfer.py", line 104, in push_testcase_to_devices_via_repl
    for path in paths:
TypeError: 'NoneType' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools\mpytool.py", line 115, in <module>
    file_transfer.push_testcase_to_devices_via_repl(file, port, tool)
TypeError: 'NoneType' object is not iterable

### Version

master (Default)

chaojibinbin, 2022-06-02:
本地更新报错

chaojibinbin, 2022-06-02:
测试使用老代码报错

skylarCai, 2022-06-02:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

zetamind, 2022-06-02:
最新的  haas studio 因为版本更新，创建的整个工程目录，代码推送目录等都有改变，建议重新创建一个新的工程，然后把老工程代码复制过去，再做推送的动作

closed 2022-06-06 by chaojibinbin