alibaba/AliOS-Things #1766: [Bug Report]:esp32-s 最新版本多线程鲍旭哦

chaojibinbin, 2022-06-02:
### What happened?

---------------------------------------------------------------log-----------------------------------------------
==== python file check /data/pyamp/main.py ====
 ==== python execute from /data/pyamp/main.py ====
helloworld
helloworld
3333
4444
11111
2222
4444
11111
helloworld
2222
3333
11111
2222
3333
4444
4444
11111

Memory allocation failed

Backtrace:0x400915c2:0x3ffcfb50 0x40091af1:0x3ffcfb70 0x4019bfde:0x3ffcfb90 0x40085537:0x3ffcfbb0 0x400926dc:0x3ffcfbe0 0x4011ccf8:0x3ffcfc20 0x4011cd59:0x3ffcfc60 0x400f9b75:0x3ffcfc80 0x400fe301:0x3ffcfcb0 0x401065c1:0x3ffcfce0 0x401066ee:0x3ffcfd00 0x40109315:0x3ffcfd20 0x400fe380:0x3ffcfdc0 0x401065c1:0x3ffcfe20 0x401065ea:0x3ffcfe40 0x401171ef:0x3ffcfe60 0x40117569:0x3ffcfef0 0x401175cd:0x3ffcff10 0x400d2fca:0x3ffcff30

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
---------------------------------------------------------------log-----------------------------------------------

---------------------------------------------------------------code-----------------------------------------------

import utime   # 延时函数在utime库中
import _thread   # 延时函数在utime库中

def theard():
    while True:
      print(11111)
      utime.sleep(1)      # 打印完之后休眠1秒

def theard1():
    while True:
      print(2222)
      utime.sleep(1)      # 打印完之后休眠1秒
def theard2():

    while True:
      print(3333)
      utime.sleep(1)      # 打印完之后休眠1秒
def theard3():

    while True:
      print(4444)
      utime.sleep(1)      # 打印完之后休眠1秒
if __name__ == '__main__':
    while True:             # 无限循环
        print("helloworld")  # 打印"helloworld"字串到console中
        utime.sleep(1)      # 打印完之后休眠1秒
        _thread.start_new_thread(theard,())
        _thread.start_new_thread(theard1,())
        _thread.start_new_thread(theard2,())
        _thread.start_new_thread(theard3,())

### Version

master (Default)

### What soultions are you seeing the problem on?

helloworld_demo

skylarCai, 2022-06-02:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

chaojibinbin, 2022-06-02:
> 请问你使用的是哪个版本的固件？你可以使用下面的命令获取版本信息： import uos uos.version()

最新的2.2版本固件。实际上2.1就是这样。2.0就不会。

zetamind, 2022-06-06:
代码看起来和你的 log 有点对应不上， 从你的代码来看，只会在进去的第一次打印 helloworld， 然后就在线程中打印， 但是 crash 的 log 好像打印了三次 helloworld 。 另外 crash 从 log 上面看是因为内存分配失败，可以 import gc，  在线程中使用 gc.mem_free() 查看当前内存剩余空间，或者加入 gc.collect() 进行内存清理。

yong171966, 2022-06-09:
>>> import uos
>>> uos.version_info()
'HaaSPython-ESP32-S3-v2.1.0-2022-05-25T14:37:27'

基于上述固件，使用issue中贴出来的代码运行，未发现crash问题。

closed 2022-06-13 by skylarCai