alibaba/AliOS-Things #1805: [Bug Report]: m5stackcore2 ota升级到最新版本后又重复下载升级

rualdeng, 2022-06-08:
### Contact Details

钉钉：rualxiao

### What happened?

A bug happened!
运行环境:
固件版本:最新版本

版本从1.1.0升级到1.2.0，后又重复下载，循环升级1.2.0
[image]

### Version

master (Default)

### What soultions are you seeing the problem on?

py_engine_esp32_demo

### Relevant log output

```shell
版本从1.1.0升级到1.2.0，后又重复下载，循环升级1.2.0
```

skylarCai, 2022-06-08:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

rualdeng, 2022-06-08:
> ```
> 感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
> 您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)
> 
> 如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。
> ```

人工支持

skylarCai, 2022-06-08:
我们尽快看看

rualdeng, 2022-06-08:
> 我们尽快看看

好的，谢谢！

rualdeng, 2022-06-08:
> 升级完成后要上报新的版本

是的，上报的是新的版本。打印的log信息就是新的版本的信息

yanxiaoyong-ch, 2022-06-08:
在云端填的什么待升级版本，升级完成重启后，第一件事就要上报在云端填写的带升级版本号

rualdeng, 2022-06-08:
[image]
这是升级1.2.0时打印的上报信息的版本号为1.1.0。升级后打印的信息的版本号为1.2.0。上报的是新版本。

rualdeng, 2022-06-08:
> 在云端填的什么待升级版本，升级完成重启后，第一件事就要上报在云端填写的带升级版本号

 云端填的版本号就是1.2.0

yanxiaoyong-ch, 2022-06-08:
ota.report要在ota.init之前执行

rualdeng, 2022-06-08:
> ota.report要在ota.init之前执行

好，我来测试一下

rualdeng, 2022-06-08:
可以了

closed 2022-06-08 by rualdeng