alibaba/AliOS-Things #1810: [Bug Report]: L298N电机驱动 驱动错误

15023019082, 2022-06-08:
### 联系方式

690482152@qq.com

### 您发现了什么问题

A bug happened!
运行环境:NodeMCU-32S开发板
固件版本:HaaSPython-ESP32-v2.2.0.bin

### 分支

master (Default)

### 请选择您发现此问题的解决方案名称

helloworld_demo

### 日志

```shell
PS E:\haas\esp32\test3\test3_demo> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools\mpytool.py -p COM8                 -f e:\haas\esp32\test3\test3_demo\src -t c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools
[*] port is  COM8
[*] Filename is  e:\haas\esp32\test3\test3_demo\src
[*] Filename is  c:\Users\Administrator\.vscode\extensions\haas.haas-studio-2.2.0\asserts\mpy_tools
C:\Users\Administrator\AppData\Roaming\Thonny
==== wait hardware reset!!! ====
==== hardware reset OK!!! ====
=== start upload... ===
=== wait python process interrupt ===
upload process...0%
upload process...57%
upload process...67%
=== upload progress end ===
3.631155252456665
1.7808268070220947
--- Miniterm on COM8  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
mount fs
 ==== python execute bootpy ====
 ==== python file check /data/pyamp/main.py ====
 ==== python execute from /data/pyamp/main.py ====
[   0.160]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[   0.160]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[   0.160]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[  10.170]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[  10.170]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[  10.170]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[  20.180]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[  20.180]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
[  20.180]<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict 
mp_task-250 158996
MicroPython v1.17-dirty on 2022-05-18T11:36:05; ESP32 module with ESP32
Type "help()" for more information.
```

skylarCai, 2022-06-08:
我们分析一下

ElliottSeer, 2022-06-08:
<E>DRIVER_PWM obj_setOption  param1 type error,param type must be dict

ElliottSeer, 2022-06-14:
Hello 问题后来解决了吗

closed 2022-07-13 by YiluMao