alibaba/AliOS-Things #1828: [Bug Report]:

123123ly, 2022-06-13:
### Contact Details

1764745743@qq.com

### What happened?

A bug happened!
运行环境:用的Haas506做的那个久坐提醒系统案例，代码串烧成功之后，网络也注册成功了，但是阿里云平台上的设备还是离线的呢？

固件版本:

### Version

master (Default)

### Relevant log output

```shell
用的Haas506做的那个久坐提醒系统案例，代码串烧成功之后，网络也注册成功了，但是阿里云平台上的设备还是离线的呢？
```

YiluMao, 2022-06-13:
好的，我们内部先看下

skylarCai, 2022-06-13:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

123123ly, 2022-06-13:
我第一次试的时候是在线的，而且在线调试在钉钉上也有消息提醒，但是后面再用的时候就不行了

wushengtws, 2022-06-13:
看你的描述是阿里云平台上的设备是离线的，感觉是和网络有关系 ，你这边可以先这样排查一下，在物联网平台创建一个产品试试，看看是不是也是第一次可以看到在线，之后就离线了

wushengtws, 2022-06-13:
比如连接上了物联网平台以后，5秒钟发一包数据看看 物联网平台的调试串口看看能不能有数据上来

123123ly, 2022-06-13:
我新建了几个，都试过的

wushengtws, 2022-06-13:
https://haas.iot.aliyun.com/solution/detail/solution?versionId=800A8E010187BE0A00000001&dataId=800A8E010187BE0A
创建物联网平台是按照这里面操作的吗

123123ly, 2022-06-13:
我用的是Haas506开发板里的久坐提醒案例

wushengtws, 2022-06-13:
我们没有HaaS506的久坐提醒案例，现在只要esp32的久坐提醒案例，请按照我上面给你的链接 中的案例操作试试看呐

YiluMao, 2022-07-13:
请问还有问题不？

YiluMao, 2022-07-15:
问题先关闭，如果有进一步问题，请reopen或者新开issue
多谢

closed 2022-07-15 by YiluMao