alibaba/AliOS-Things #1836: 不能上传PYTHON程序

712163, 2022-06-16:
### What happened?

运行环境: WIN10 NodeMCU-ESP-S3-32S-Kit
固件版本:HaaSPython-ESP-S3-12K-Kit-v2.2.0.bin

不能上传PYTHON程序。怀疑是串口波特率问题，在UPGRADE.JSON 改端口为115200后上传系统自动改回15000000

### Version

master (Default)

### Relevant log output

```shell
Windows PowerShell
版权所有 (C) Microsoft Corporation。保留所有权利。

尝试新的跨平台 PowerShell https://aka.ms/pscore6

PS C:\NIJI\project\ESP32\print-escpos-python\pr> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\71216\.vscode\extensions\haas.haas-studio-2.2.0\asserts\burntool\transymodem.py -d COM7       
          -b 1500000 c:\NIJI\project\ESP32\print-escpos-python\pr\upgrade\pyamp.zip
Namespace(baudrate='1500000', device='COM7', files=['c:\\NIJI\\project\\ESP32\\print-escpos-python\\pr\\upgrade\\pyamp.zip'])
the filepath is abs path
serial port is COM7
transfer baudrate is 1500000
filepath is c:\NIJI\project\ESP32\print-escpos-python\pr\upgrade\pyamp.zip
b''
Please reboot the board manually.
Please reboot the board manually, and try it again.
PS C:\NIJI\project\ESP32\print-escpos-python\pr>
```

skylarCai, 2022-06-16:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

senwangal, 2022-06-16:
创建项目时，选错了板卡，按照你的固件名字，推测你应该是ESP-S3-12K的板卡，创建项目时请选择这个板卡

senwangal, 2022-06-16:
[image]

712163, 2022-06-16:
Windows PowerShell
版权所有 (C) Microsoft Corporation。保留所有权利。

尝试新的跨平台 PowerShell https://aka.ms/pscore6

PS C:\NIJI\project\ESP32\print-escpos-python\pr> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\71216\.vscode\extensions\haas.haas-studio-2.2.0\asserts\burntool\transymodem.py -d COM7                 -b 1500000 c:\NIJI\project\ESP32\print-escpos-python\pr\upgrade\pyamp.zip
Namespace(baudrate='1500000', device='COM7', files=['c:\\NIJI\\project\\ESP32\\print-escpos-python\\pr\\upgrade\\pyamp.zip'])
the filepath is abs path
serial port is COM7
transfer baudrate is 1500000
filepath is c:\NIJI\project\ESP32\print-escpos-python\pr\upgrade\pyamp.zip
b''
Please reboot the board manually.
Please reboot the board manually, and try it again.
PS C:\NIJI\project\ESP32\print-escpos-python\pr> 

新建项目，再次选板子，一样的报错

senwangal, 2022-06-16:
麻烦提供下根目录src下board.json的文件内容

712163, 2022-06-16:
> 麻烦提供下根目录src下board.json的文件内容

{
    "name": "esp32-s3",
    "version": "1.0.0",
    "io": {
    },
    "debugLevel": "ERROR",
    "repl": "disable"
  }

712163, 2022-06-16:
已经解决

解决方法：
.vscode/settings.json 配置中添加一项 "haasStudio.hardware.board": "nodemcu32s",

closed 2022-06-16 by 712163

senwangal, 2022-06-16:
创建项目需要确认用期望的板卡
.vscode/settings.json 配置中添加一项 "haasStudio.hardware.board": "nodemcu32s",也是受此影响

[image]