alibaba/AliOS-Things #1903: Haas edu k1 显示屏不亮

pikapiya, 2022-07-07:
### Contact Details

lyg301@163.com

### What do you want?

板子没有摔，可以正常烧录程序，也能运行结果，但显示屏就是没有反应，不烧程序，显示屏开机和复位也不会有显示

### Version

master (Default)

skylarCai, 2022-07-07:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

pikapiya, 2022-07-07:
> ```
> 感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
> 您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)
> 
> 如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。
> ```

人工支持

YiluMao, 2022-07-13:
请问之前是显示的是吗？
有没有更新过固件
可以在我们HaaS官网上下载最新固件试一下
https://haas.iot.aliyun.com/haasapi/index.html?#/Python/docs/zh-CN/startup/startup

YiluMao, 2022-07-15:
问题先关闭，如果有进一步问题，请reopen或者新开issue
多谢

closed 2022-07-15 by YiluMao