alibaba/AliOS-Things #1906: [Bug Report]:

e3control-zhanglimin, 2022-07-11:
### 联系方式

liming.zhang@の

### 您发现了什么问题

Aos激活失败

### 分支

rel_3.3.0 (Main)

### 请选择您发现此问题的解决方案名称

amp_demo

### 日志

```shell
正在安装 aos, 请等待...
[1/4] 安装 python py3tftp
C:\ProgramData\aos\miniconda3\Scripts\pip.exe install -i https://pypi.tuna.tsinghua.edu.cn/simple --trusted-host pypi.tuna.tsinghua.edu.cn py3tftp
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: py3tftp in c:\programdata\aos\miniconda3\lib\site-packages (1.3.0)
[2/4] 安装 python pyclean
C:\ProgramData\aos\miniconda3\Scripts\pip.exe install -i https://pypi.tuna.tsinghua.edu.cn/simple --trusted-host pypi.tuna.tsinghua.edu.cn pyclean
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: pyclean in c:\programdata\aos\miniconda3\lib\site-packages (2.2.0)
[3/4] 安装 python pyserial
C:\ProgramData\aos\miniconda3\Scripts\pip.exe install -i https://pypi.tuna.tsinghua.edu.cn/simple --trusted-host pypi.tuna.tsinghua.edu.cn pyserial
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: pyserial in c:\programdata\aos\miniconda3\lib\site-packages (3.5)
[4/4] 安装 HaaS Python类库
安装 HaaS Python类库 失败：
Request failed with status code 403，点左下角"激活AOS插件"重试.正在安装 aos, 请等待...
[1/4] 安装 python py3tftp
C:\ProgramData\aos\miniconda3\Scripts\pip.exe install -i https://pypi.tuna.tsinghua.edu.cn/simple --trusted-host pypi.tuna.tsinghua.edu.cn py3tftp
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: py3tftp in c:\programdata\aos\miniconda3\lib\site-packages (1.3.0)
[2/4] 安装 python pyclean
C:\ProgramData\aos\miniconda3\Scripts\pip.exe install -i https://pypi.tuna.tsinghua.edu.cn/simple --trusted-host pypi.tuna.tsinghua.edu.cn pyclean
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: pyclean in c:\programdata\aos\miniconda3\lib\site-packages (2.2.0)
[3/4] 安装 python pyserial
C:\ProgramData\aos\miniconda3\Scripts\pip.exe install -i https://pypi.tuna.tsinghua.edu.cn/simple --trusted-host pypi.tuna.tsinghua.edu.cn pyserial
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: pyserial in c:\programdata\aos\miniconda3\lib\site-packages (3.5)
[4/4] 安装 HaaS Python类库
安装 HaaS Python类库 失败：
Request failed with status code 403，点左下角"激活AOS插件"重试.
```

YiluMao, 2022-07-13:
请点击激活试试

YiluMao, 2022-07-15:
问题先关闭，如果有进一步问题，请reopen或者新开issue
多谢

closed 2022-07-15 by YiluMao