alibaba/AliOS-Things #1908: Haas edu k1 的陀螺仪模块mpu6050,有python案例么,board.json里是这么添加的么,不知道怎么调用它

2323623353, 2022-07-13:
### Contact Details

2323623353@qq.com

### What do you want?

[image]

### Version

master (Default)

rockzhouchina, 2022-07-14:
Contact Details

YiluMao, 2022-07-14:
https://github.com/alibaba/AliOS-Things/tree/master/haas_lib_bundles/python/libraries/mpu6050
您可以参考下这个文档

2323623353, 2022-07-14:
> https://github.com/alibaba/AliOS-Things/tree/master/haas_lib_bundles/python/libraries/mpu6050 您可以参考下这个文档
我试过,刷入进去,串口都没输出,没反应

2323623353, 2022-07-14:
我应该知道问题了,这个k1有两个版本咋也不说明,还是我才找到,官方说明书也不说

2323623353, 2022-07-14:
我知道问题了,k1还有两个版本,官网也不说明,我的是qmi8610,这还是我搜半天找到你们的文档,qmi这个传感器我没找到调用方法,有说明文档么

YiluMao, 2022-07-15:
qmi这个文档确实没找到
不过你可以参考下这个库实现
https://github.com/alibaba/AliOS-Things/blob/master/haas_lib_bundles/python/libraries/qmi8610/qmi8610.py

skylarCai, 2022-07-19:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-07-20:
请问还有什么问题吗？