alibaba/AliOS-Things #1914: [Bug Report]: 串口无法连接

emo666, 2022-07-15:
### Contact Details

320613249@qq.com

### What happened?

A bug happened!
运行环境: win10
固件版本: v2.2.
[image]
0

### Version

rel_3.3.0 (Main)

### Relevant log output

```shell
PowerShell 7.2.5
Copyright (c) Microsoft Corporation.

https://aka.ms/powershell
Type 'help' to get help.

PS F:\ailiyun\test_demo> "C:\ProgramData\aos\miniconda3\condabin\conda.bat" activate C:\ProgramData\aos\miniconda3,set PATH=C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;%PATH%&python -m serial.tools.miniterm COM6 115200 --raw
ParserError: 
Line |
   1 |  "C:\ProgramData\aos\miniconda3\condabin\conda.bat" activate C:\Progra …
     |                                                     ~~~~~~~~
     | Unexpected token 'activate' in expression or statement.
```

YiluMao, 2022-07-18:
我们先看看

Virace, 2022-12-08:
所以是这个问题7月份就有了，现在还这样