alibaba/AliOS-Things #1925: [Bug Report]: vscode运行只有“本地更新”

weipx666, 2022-07-23:
### Contact Details

439451902@qq.com

### What happened?

运行环境: windows、vscode
固件版本: HaaSPython-HaaSEDUK1-v2.2.0.bin
haas-studio版本: 2.3.1

[image]

### Version

master (Default)

### What soultions are you seeing the problem on?

eduk1_demo

### Relevant log output

```shell
-
```

skylarCai, 2022-07-23:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

sunlrain, 2022-07-27:
对，目前在本地更新上遇到什么问题吗？

weipx666, 2022-07-27:
> 对，目前在本地更新上遇到什么问题了吗？

本地更新没有问题, 想弄 在线更新,找不到入口

YiluMao, 2022-08-02:
目前这个功能已经下线了
请问您对这个功能的需求是什么？

YiluMao, 2022-08-03:
请问还有问题不？

YiluMao, 2022-08-04:
先关闭，有问题请再提问

closed 2022-08-04 by YiluMao