alibaba/AliOS-Things #1928: [Feature Request]:

xiamichen, 2022-08-02:
### 联系方式

1757789374@qq.com

### 新功能描述

[明栈 M5StackCore2和M5stack Core使用相同的ESP32芯片、flash和LCD屏幕，那Core2的HAAS python固件能否直接烧录至M5stack core中使用
[image]
[image]

### 分支

master (Default)

skylarCai, 2022-08-02:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-08-03:
我们没有验证过，可以尝试烧录下看看效果

YiluMao, 2022-08-04:
请问还有问题不？

YiluMao, 2022-08-05:
先关闭，有问题再开

closed 2022-08-05 by YiluMao

xiamichen, 2022-10-11:
没有了，我购买的相关硬件在路上，我到时候烧录一下看看情况

YiluMao, 2022-10-11:
ok