alibaba/AliOS-Things #1929: haas ESP32  I2C 适配问题

pufangwen, 2022-08-04:
### Contact Details

pfwen@139.com

### What do you want?

我想咨询下 我目前用的是esp32 做Haas开发，但是需要 自定义I2C的引脚，请问在borad.json中如何配置J2C的引脚，还是已经封闭必须是GPIO 21 22
[image]

### Version

master (Default)

skylarCai, 2022-08-04:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-09-13:
如果是这个板子，硬件决定了只能是这两个引脚，请参考文档：
https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16145223.0.0.595660b1siyJSq#/Python/docs/zh-CN/haas_extended_api/driver/i2c