alibaba/AliOS-Things #1932: [Bug Report]: 点击激活aos没有反应，重启vscode也不行

lonngxiang, 2022-08-07:
### Contact Details

1083733327@qq.com

### What happened?

A bug happened!
运行环境:
固件版本:esp 2.1.0

### Version

master (Default)

skylarCai, 2022-08-07:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-08-08:
之前是好的吗？

lonngxiang, 2022-08-08:
> 之前是好的吗？ 是最近升级造成的吗？

怎么说，最开始安装插件是成功的，但是创建项目点击部署一直没有反应；然后重启vscode后就显示激活aos，但点击一直没反应，后来看了下不管这个直接创建项目又可以

lonngxiang, 2022-08-08:
> 之前是好的吗？ 是最近升级造成的吗？

我后面直接用别的ide  thony去操作没问题

YiluMao, 2022-08-09:
好的，如果有进一步问题，请再提问
这个问题我先关闭了哈

closed 2022-08-09 by YiluMao