alibaba/AliOS-Things #1933: [Bug Report]: ESP32车辆定位项目中无法获取GPS模块数据，一直显示[12658.727]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;

Sssyuan, 2022-08-16:
### 联系方式

2432552396@qq.com

### 您发现了什么问题

A bug happened!
运行环境:PS D:\ESP32\workspace\dingwei> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\86159\.vscode\extensions\haas.haas-studio-2.3.1\asserts\mpy_tools\mpytool.py -p COM5                 -f d:\ESP32\workspace\dingwei\src -t c:\Users\86159\.vscode\extensions\haas.haas-studio-2.3.1\asserts\mpy_tools -ou（不知道是不是这个）
固件版本:HaaSPython-ESP32-v1.0.2.bin

已经确定能够联网连云端，使用串口助手查看定位模块时也能获取信息，就是ESP32无法获取定位模块数据，一直显示
<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;

我ESP32是ESP32 DEVKITV1 TX0对应GPIO01 RX0对应GPIO03

board.json
{
  "name": "esp32",
  "version": "1.0.0",
  "io": {
    "gnss": {
      "type": "UART",
      "port": 5,
      "dataWidth": 8,
      "baudRate": 9600,
      "stopBits": 1,
      "flowControl": "disable",
      "parity": "none"
    }
  },
  "debugLevel": "ERROR",
  "repl": "ensable"
}

### 分支

master (Default)

### 请选择您发现此问题的解决方案名称

helloworld_demo

### 日志

```shell
VSCODE上（正常）：
PS D:\ESP32\workspace\dingwei> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\86159\.vscode\extensions\haas.haas-studio-2.3.1\asserts\mpy_tools\mpytool.py -p COM5                 -f d:\ESP32\workspace\dingwei\src -t c:\Users\86159\.vscode\extensions\haas.haas-studio-2.3.1\asserts\mpy_tools -ou
[*] port is  COM5
[*] Filename is  d:\ESP32\workspace\dingwei\src
[*] Filename is  c:\Users\86159\.vscode\extensions\haas.haas-studio-2.3.1\asserts\mpy_tools
C:\Users\86159\AppData\Roaming\Thonny
==== wait hardware reset!!! ====
==== hardware reset OK!!! ====
=== start upload... ===
=== wait python process interrupt ===
upload process...0%
upload process...6%
upload process...9%
upload process...12%
upload process...15%
upload process...20%
upload process...23%
upload process...26%
upload process...29%
upload process...34%
upload process...37%
upload process...40%
upload process...42%
upload process...45%
upload process...48%
upload process...50%
upload process...53%
upload process...56%
upload process...59%
upload process...61%
upload process...64%
upload process...67%
upload process...70%
upload process...72%
upload process...75%
upload process...78%
upload process...81%
upload process...83%
upload process...86%
upload process...89%
upload process...92%
upload process...94%
upload process...97%
upload process...100%
=== upload progress end ===

使用Thonny查看时
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
wifi_connected: False
('172.20.10.11', '255.255.255.240', '172.20.10.1', '172.20.10.1')
establish tcp connection with server(host='hjuaCPwFqyH.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
sleep for 1 s
tcp_connect: can only connect from state CLOSED
success to establish tcp, fd=54
物联网平台连接成功
sleep for 2s
[  53.880]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;
[0.0, 'W'] [0.0, 'N'] 0.0
[  55.880]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;
[0.0, 'W'] [0.0, 'N'] 0.0
[  57.880]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;
[0.0, 'W'] [0.0, 'N'] 0.0
[  59.890]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;
[0.0, 'W'] [0.0, 'N'] 0.0
[  61.890]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;
[0.0, 'W'] [0.0, 'N'] 0.0
[  63.890]<E>DRIVER_UART obj_read: py_board_get_node_by_handle failed;
[0.0, 'W'] [0.0, 'N'] 0.0
```

skylarCai, 2022-08-16:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-09-13:
1. 请更新固件到最新试试
2. 请用串口助手查一下ESP32的串口有没有数据