alibaba/AliOS-Things #1939: [Bug Report]: ESP32 Uart 读写无数据

e3control-zhanglimin, 2022-09-06:
### 联系方式

limin.zhang@e3control.cn

### 您发现了什么问题

A bug happened!
运行环境:ESP32
固件版本: HaaSPython-ESP32-v2.2.0.bin

{
    "name": "esp32",
    "version": "1.0.0",
    "io": {
      "gnss": {
        "type": "UART",
        "port": 2,
        "dataWidth": 8,
        "baudRate": 115200,
        "stopBits": 1,
        "flowControl": "disable",
        "parity": "none"
      }
    },
    "debugLevel": "DEBUG",
    "repl": "disable"
  }
[image]

### 分支

master (Default)

### 请选择您发现此问题的解决方案名称

helloworld_demo

skylarCai, 2022-09-06:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-09-06:
我们内部测试下

YiluMao, 2022-09-06:
您好，我这边测试了下我们的“车辆定位系统”案例
https://haas.iot.aliyun.com/solution/detail/solution?versionId=800AC00BDFA5C1AB00000002&dataId=800AC00BDFA5C1AB
这个是用uart的，结果是好的

要不您先测试下这个案例，看看有没有问题？

e3control-zhanglimin, 2022-09-08:
我这边用定位的demo，使用一个usb转ttl，一边连接esp32 的rx和tx，一边在pc上查看接收数据，通过esp32 的uart写的数据pc这边还是收不到
[image]
[image]
[image]
[image]
[image]
[image]

e3control-zhanglimin, 2022-09-08:
以下是接线图：
[image]
[image]
[image]
[image]

YiluMao, 2022-09-08:
我看您修改了案例里面的uart的配置
能帮忙先改回去，再修改PC端的配置，试一下先？

e3control-zhanglimin, 2022-09-08:
改回案例中的默认Uart配置也是不行，同时我这边又找了一个新的板子，也是同样的问题，能发给我们一份您测试用的固件程序吗，我这边用您的固件再测试一下

e3control-zhanglimin, 2022-09-09:
[gnss2_demo.zip](https://github.com/alibaba/AliOS-Things/files/9531386/gnss2_demo.zip)
这个是我这边的工程，您帮忙试试Uart有输出没

YiluMao, 2022-09-13:
上面的板子不是我们推荐的NodeMCU，接线不能完全参考我们的文档
请检查后再试试

closed 2022-09-13 by YiluMao