alibaba/AliOS-Things #1940: [Bug Report]: esp32 连接iotDevice 10分左右断开iot连接

e3control-zhanglimin, 2022-09-06:
### 联系方式

limin.zhang@e3control.cn

### 您发现了什么问题

A bug happened!
运行环境:Esp32
固件版本:HaaSPython-ESP32-v2.2.0.bin
[image]
[image]

### 分支

master (Default)

### 请选择您发现此问题的解决方案名称

helloworld_demo

### 日志

```shell
props report success
1970-01-01 08:07:12 <I> loop cnt 421 
1970-01-01 08:07:13 <I> loop cnt 422 
1970-01-01 08:07:14 <I> loop cnt 423 
1970-01-01 08:07:15 <I> loop cnt 424 
1970-01-01 08:07:16 <I> loop cnt 425 
1970-01-01 08:07:17 <I> loop cnt 426 
1970-01-01 08:07:18 <I> loop cnt 427 
1970-01-01 08:07:19 <I> loop cnt 428 
1970-01-01 08:07:20 <I> loop cnt 429 
1970-01-01 08:07:21 <I> loop cnt 430 
1970-01-01 08:07:22 <I> loop cnt 431 
1970-01-01 08:07:23 <I> loop cnt 432 
1970-01-01 08:07:24 <I> loop cnt 433 
1970-01-01 08:07:25 <I> loop cnt 434 
1970-01-01 08:07:26 <I> loop cnt 435 
1970-01-01 08:07:27 <I> loop cnt 436 
1970-01-01 08:07:28 <I> loop cnt 437 
1970-01-01 08:07:29 <I> loop cnt 438 
1970-01-01 08:07:30 <I> loop cnt 439 
1970-01-01 08:07:31 <I> loop cnt 440 
1970-01-01 08:07:32 <I> loop cnt 441 
1970-01-01 08:07:33 <I> loop cnt 442 
1970-01-01 08:07:34 <I> loop cnt 443 
1970-01-01 08:07:35 <I> loop cnt 444 
1970-01-01 08:07:36 <I> loop cnt 445 
1970-01-01 08:07:37 <I> loop cnt 446 
1970-01-01 08:07:38 <I> loop cnt 447 
1970-01-01 08:07:39 <I> loop cnt 448 
1970-01-01 08:07:40 <I> loop cnt 449 
1970-01-01 08:07:41 <I> loop cnt 450 
1970-01-01 08:07:42 <I> loop cnt 451 
1970-01-01 08:07:43 <I> loop cnt 452 
1970-01-01 08:07:44 <I> loop cnt 453 
1970-01-01 08:07:45 <I> loop cnt 454 
1970-01-01 08:07:46 <I> loop cnt 455 
1970-01-01 08:07:47 <I> loop cnt 456 
1970-01-01 08:07:48 <I> loop cnt 457 
1970-01-01 08:07:49 <I> loop cnt 458 
1970-01-01 08:07:50 <I> loop cnt 459 
1970-01-01 08:07:51 <I> loop cnt 460 
1970-01-01 08:07:52 <I> loop cnt 461 
1970-01-01 08:07:53 <I> loop cnt 462 
1970-01-01 08:07:54 <I> loop cnt 463 
1970-01-01 08:07:55 <I> loop cnt 464 
1970-01-01 08:07:56 <I> loop cnt 465 
1970-01-01 08:07:57 <I> loop cnt 466 
[... 166 lines cut ...]
1970-01-01 08:10:26 <I> loop cnt 611 
1970-01-01 08:10:27 <I> loop cnt 612 
1970-01-01 08:10:28 <I> loop cnt 613 
1970-01-01 08:10:29 <I> loop cnt 614 
fail to establish tcp
1970-01-01 08:10:30 <I> loop cnt 615 
establish tcp connection with server(host='a1z5POtAhDY.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
1970-01-01 08:10:31 <I> loop cnt 616 
1970-01-01 08:10:32 <I> loop cnt 617 
1970-01-01 08:10:33 <I> loop cnt 618 
1970-01-01 08:10:34 <I> loop cnt 619 
1970-01-01 08:10:35 <I> loop cnt 620 
1970-01-01 08:10:36 <I> loop cnt 621 
1970-01-01 08:10:37 <I> loop cnt 622 
1970-01-01 08:10:38 <I> loop cnt 623 
1970-01-01 08:10:39 <I> loop cnt 624 
1970-01-01 08:10:40 <I> loop cnt 625 
1970-01-01 08:10:41 <I> loop cnt 626 
1970-01-01 08:10:42 <I> loop cnt 627 
1970-01-01 08:10:43 <I> loop cnt 628 
1970-01-01 08:10:44 <I> loop cnt 629 
fail to establish tcp
1970-01-01 08:10:45 <I> loop cnt 630 
establish tcp connection with server(host='a1z5POtAhDY.iot-as-mqtt.cn-shanghai.aliyuncs.com', port=[443])
```

skylarCai, 2022-09-06:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-09-13:
请问这个问题还在不？

e3control-zhanglimin, 2022-09-14:
这个问题还在，下面是我的demo和日志
[esp32_demo.zip](https://github.com/alibaba/AliOS-Things/files/9563539/esp32_demo.zip)
[log.txt](https://github.com/alibaba/AliOS-Things/files/9563551/log.txt)