alibaba/AliOS-Things #1941: [Bug Report]:

15023019082, 2022-09-09:
### 联系方式

690482152@qq.com

### 您发现了什么问题

A bug happened!
运行环境:NodeMCU-32S开发板
固件版本:HaaSPython-ESP32-v2.2.0
mqtt自动重连失效，获取连接状态方法get_mqttsta报错

[get_mqttsta - 获取MQTT的连接状态](https://haas.iot.aliyun.com/haasapi/index.html?spm=a2cpu.b16834026.0.0.251c60b1jkYCmP#/Python/docs/zh-CN/standard_api/umqtt?id=get_mqttsta-%e8%8e%b7%e5%8f%96mqtt%e7%9a%84%e8%bf%9e%e6%8e%a5%e7%8a%b6%e6%80%81)
函数原型：

MQTTClient.get_mqttsta()

参数说明：
无

返回值

0：连接成功
1：连接中
2：服务端连接关闭
-1：连接异常

调用disconnect之后，再调用get_mqttsta返回-1

### 分支

master (Default)

### 请选择您发现此问题的解决方案名称

helloworld_demo

### 日志

```shell
Traceback (most recent call last):
  File "/data/pyamp/main.py", line 99, in <module>
AttributeError: 'MQTTClient' object has no attribute 'get_mqttsta'
mp_task-258 178404
MicroPython v1.17-dirty on 2022-06-21T00:50:31; ESP32 module with ESP32
Type "help()" for more information.
```

skylarCai, 2022-09-09:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

YiluMao, 2022-09-13:
能帮忙把您的代码文件贴上来吗？我这边可以试一下

15023019082, 2022-09-15:
1

closed 2022-09-15 by 15023019082