alibaba/AliOS-Things #1945: COM3端口访问失败

Zhao-Dream, 2022-10-03:
### Contact Details

2189885488@qq.com

### What happened?

刚到的HaaS Edu K1-烧录固件无法访问COM端口
运行环境: win11
固件版本:1.1.0

### Version

master (Default)

### Relevant log output

```shell
PS D:\VSCode\Project\测试\test_demo> $Env:Path="C:\ProgramData\aos\miniconda3;C:\ProgramData\aos\miniconda3\Scripts;C:\ProgramData\aos\miniconda3\Library\bin;$Env:Path";python c:\Users\Zhaomeng\.vscode\extensions\haas.haas-studio-2.3.1\asserts\burntool\haaseduk1\flash_program.py --bin=d:\新建文件夹\HaaSPython-HaaSEDUK1-v2.2.0\HaaSPython-HaaSEDUK1-v2.2.0\HaaSPython-HaaSEDUK1-v2.2.0.bin --bin=d:\新建文件夹\HaaSPython-HaaSEDUK1-v2.2.0\HaaSPython-HaaSEDUK1-v2.2.0\littlefs.bin#0xB32000 --port=COM3                
binary file is ['d:\\新建文件夹\\HaaSPython-HaaSEDUK1-v2.2.0\\HaaSPython-HaaSEDUK1-v2.2.0\\HaaSPython-HaaSEDUK1-v2.2.0.bin', 'd:\\新建文件夹\\HaaSPython-HaaSEDUK1-v2.2.0\\HaaSPython-HaaSEDUK1-v2.2.0\\littlefs.bin#0xB32000']
serial port is COM3
bin_files is  [('d:\\新建文件夹\\HaaSPython-HaaSEDUK1-v2.2.0\\HaaSPython-HaaSEDUK1-v2.2.0\\HaaSPython-HaaSEDUK1-v2.2.0.bin', '0'), ('d:\\新建文件夹\\HaaSPython-HaaSEDUK1-v2.2.0\\HaaSPython-HaaSEDUK1-v2.2.0\\littlefs.bin', '0xB32000')]
Traceback (most recent call last):
  File "c:\Users\Zhaomeng\.vscode\extensions\haas.haas-studio-2.3.1\asserts\burntool\haaseduk1\flash_program_ll.py", line 103, in burn_bin_files
    serialport.open()
  File "C:\ProgramData\aos\miniconda3\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, '拒绝访问。', None, 5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\Zhaomeng\.vscode\extensions\haas.haas-studio-2.3.1\asserts\burntool\haaseduk1\flash_program.py", line 121, in <module>
    main()
  File "c:\Users\Zhaomeng\.vscode\extensions\haas.haas-studio-2.3.1\asserts\burntool\haaseduk1\flash_program.py", line 118, in main
    burn_bin_files(myconfig["serialport"], myconfig['baudrate'], bin_files)
  File "c:\Users\Zhaomeng\.vscode\extensions\haas.haas-studio-2.3.1\asserts\burntool\haaseduk1\flash_program_ll.py", line 105, in burn_bin_files
    raise Exception("Failed to open serial port: %s!" % portnum)
Exception: Failed to open serial port: COM3!
```

skylarCai, 2022-10-03:
感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
    您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)

    如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。

Zhao-Dream, 2022-10-04:
人工支持

Zhao-Dream, 2022-10-04:
> ```
> 感谢您提出的宝贵问题，我们有7x24小时服务的“HaaS百事通”客服系统，
> 您可以先尝试能否解决您的问题(https://haas.iot.aliyun.com/?ask=1&f=a2cre.b82925042)
> 
> 如果“HaaS百事通”没有解决您的问题，请回复“人工支持”，我们会在工作日(10:00-12:00/14:00-18:00)10分钟内回复您。
> ```

人工支持

YiluMao, 2022-10-08:
帮忙确定下，设备被系统识别的串口是COM3吗？
另外，如果是的话， 有没有别的工具占用了COM3？

Zhao-Dream, 2022-10-08:
> 帮忙确定下，设备被系统识别的串口是COM3吗？ 另外，如果是的话， 有没有别的工具占用了COM3？

是 已多次尝试插拔设备 确定是COM3 无任何工具占用COM3 操作包括重新设置端口为其他端口号并重启电脑立即烧录