alibaba/AliOS-Things #1960: [Bug Report]: HaaS EDU K1无法显示网络请求的中文字符串

makelove, 2023-01-04:
### Contact Details

微信:sexy8dream

### What happened?

A bug happened!
运行环境:windows 10
固件版本:HaaSPython-HaaSEDUK1-v2.2.0

把main.py转换为gbk格式文件
使用framebuf可以显示py程序里定义好的中文
但 通过网络请求拿到的中文字符串，无法显示，utf-8编码
把它转成gbk格式编码，也无法显示

代码是alcohol_detect酒精检测的例子里的
```python
# OLED初始化
def oledInit():
    global oled

    # 字库文件存放于项目目录 font, 注意若用到了中英文字库则都需要放置
    framebuf.set_font_path(framebuf.FONT_ASC12_8, '/data/font/ASC12_8')
    framebuf.set_font_path(framebuf.FONT_ASC16_8, '/data/font/ASC16_8')
    framebuf.set_font_path(framebuf.FONT_ASC24_12, '/data/font/ASC24_12')
    framebuf.set_font_path(framebuf.FONT_ASC32_16, '/data/font/ASC32_16')

    oled_spi = SPI()
    oled_spi.open("oled_spi")

    oled_res = GPIO()
    oled_res.open("oled_res")

    oled_dc = GPIO()
    oled_dc.open("oled_dc")

    #oled像素132*64
    oled = sh1106.SH1106_SPI(132, 64, oled_spi, oled_dc, oled_res)

# OLED显示
# text:显示的文本
# x:水平坐标 y:垂直坐标
# color:颜色
# clear: True-清屏显示 False-不清屏显示
# sz:字体大小
def oledShowText(text, x, y, color, clear, sz):
    global oled
    if clear:
        oled.fill(0) # 清屏
    oled.text(text, x, y, color, size = sz)
    oled.show()
```
oled.text(text, x, y, color, size = sz)
字体size这个参数无法使用
framebuf的text函数，没有size参数

### Version

master (Default)

### Relevant log output

```shell
一堆bug
文档不全
api参数都没写清楚，叫人怎样写代码？
```

makelove, 2023-01-06:
解决
[【乱码】HaaS EDU K1 开发板 不能正常显示 中文，utf-8 乱码](https://www.bilibili.com/video/BV1t84y1e76u/)

YiluMao, 2023-01-06:
帅
要不把解决方案PR提上来？

closed 2023-01-09 by YiluMao